ROVI-org/battery-data-toolkit #134: Unable to add cycle_stats to existing data

victorventuri, 2025-05-01:
I am using `data = BatteryDataset.from_hdf(...)` to load in synthetic data. The raw data is in `data.tables['raw_data']`, but there is no `cycle_stats` table. When I attempt to add that using `CapacityPerCyle().add_summaries(data=data)`, I get the error 
```python
AttributeError: 'BatteryDataset' object has no attribute 'cycle_stats'
```
Should I generate the `'cycle_stats'` table prior to calling the `CapacityPerCycle` summarizer? If so, what other tool should I use for that?

WardLT, 2025-05-01:
> <https://github.com/ROVI-org/battery-data-toolkit/issues/134> to @WardLT

This is a confusing part of battery-data-toolkit which I need to adjust.

Try loading using the "CellDataset" instead

```
from battdat.data import CellDataset

data = CellDataset.from_hdf(
```

victorventuri, 2025-05-01:
Nice, seems to work, but I get a lot of 
```python
UserWarning: Unable to clearly detect if battery started charged or discharged in cycle 1018. Amount discharged is 7.32e+04 A-s, charged is 7.33e+04 A-s
  warnings.warn(f'Unable to clearly detect if battery started charged or discharged in cycle {cyc}. '
```
as it goes through the entire dataset. Not a big deal, just letting you know.

Going to keep this open for now to help remind you about adjusting maybe the tutorials or something of that nature to avoid confusion.

closed 2025-05-05 by WardLT